OpenDriveLab/TCP #11: Will the pretrained model weights for TCP be released?

Kait0, 2022-10-05:
Hi,

thank you for releasing the code and dataset.
Will you also release the pretrained weights of your model that you have used for your leaderboard submission?
This would make it easier for people to reproduce the results/evaluate TCP on other benchmarks.

JIANGZhengm, 2022-10-22:
> Since there are some small modifications, we will re-train the model for our validation experiments and release it.
Looking forward to your Baidu cloud connection and pre-trained weights!

raozhongyu, 2022-10-26:
Thanks for your great work. Did the pre-trained weights released?

Kin-Zhang, 2022-10-26:
If you like, I can upload the model I trained in the whole dataset as the paper set. More discussion posts I publish here and screenshot here: [https://github.com/Kin-Zhang/LAV/discussions/7](https://github.com/Kin-Zhang/LAV/discussions/7)

And the WPH-commit said I believe the next version after modification would be more excellent.

raozhongyu, 2022-10-26:
> 
Thanks a lot.

raozhongyu, 2022-10-26:
> If you like, I can upload the model I trained in the whole dataset as the paper set. More discussion posts I publish here and screenshot here: [Kin-Zhang/LAV#7](https://github.com/Kin-Zhang/LAV/discussions/7)
> 
> And the WPH-commit said I believe the next version after modification would be more excellent.
Thanks a lot.

Kin-Zhang, 2022-10-26:
Here is the [onedrive download link](https://hkustconnect-my.sharepoint.com/:u:/g/personal/qzhangcb_connect_ust_hk/ETe4OX79JldIsvgNGIs5NmIBkJnfPDG2G5Ij-8ha518rOw?e=BG1dek) for model I trained in the whole dataset as the discussion said above.

raozhongyu, 2022-10-26:
> Here is the [onedrive download link](https://hkustconnect-my.sharepoint.com/:u:/g/personal/qzhangcb_connect_ust_hk/ETe4OX79JldIsvgNGIs5NmIBkJnfPDG2G5Ij-8ha518rOw?e=BG1dek) for model I trained in the whole dataset as the discussion said above.
Thanks a lot!

a1wj1, 2023-01-08:
> If you like, I can upload the model I trained in the whole dataset as the paper set. More discussion posts I publish here and screenshot here: [Kin-Zhang/carla-expert#2](https://github.com/Kin-Zhang/carla-expert/discussions/2)
> 
> As WPH-commit said, I believe the next version after modification would be more excellent.
Hello, I used your ckpt model file and reported the following error：
Skipping scenario 'Scenario4' due to setup error: Error: Unable to spawn vehicle vehicle.diamondback.century at Transform
Could you tell me why? Thank you！

sqb2145, 2023-03-27:
> Here is the [onedrive download link](https://hkustconnect-my.sharepoint.com/:u:/g/personal/qzhangcb_connect_ust_hk/ETe4OX79JldIsvgNGIs5NmIBkJnfPDG2G5Ij-8ha518rOw?e=BG1dek) for model I trained in the whole dataset as the discussion said above.

Hi I'm having some trouble running the evaluation file. Can you please tell me what am I doing wrong here?
Here is the error that I run into:

> (TCP) psu@psu-Serval:~/TCP$ sh leaderboard/scripts/run_evaluation.sh
init statistics_manager
begin
> Registering the global statistics
Exception ignored in: <function LeaderboardEvaluator.__del__ at 0x7fa81f4b5560>
Traceback (most recent call last):
  File "leaderboard/leaderboard/leaderboard_evaluator.py", line 150, in __del__
    self._cleanup()
  File "leaderboard/leaderboard/leaderboard_evaluator.py", line 183, in _cleanup
    self._agent_watchdog.stop()
  File "/home/psu/TCP/scenario_runner/srunner/scenariomanager/watchdog.py", line 72, in stop
    self._timer.cancel()
AttributeError: 'NoneType' object has no attribute 'cancel'

And this is what my run_evaluation.sh file looks like:

`#!/bin/bash
#export CARLA_ROOT= PATH_TO_CARLA
#export CARLA_SERVER=${CARLA_ROOT}/CarlaUE4.sh
#export PYTHONPATH=$PYTHONPATH:${CARLA_ROOT}/PythonAPI
#export PYTHONPATH=$PYTHONPATH:${CARLA_ROOT}/PythonAPI/carla
export PYTHONPATH=$PYTHONPATH:/home/psu/carla/PythonAPI/carla/dist/carla-0.9.10-py3.7-linux-x86_64.egg
export PYTHONPATH=$PYTHONPATH:leaderboard
export PYTHONPATH=$PYTHONPATH:leaderboard/team_code
export PYTHONPATH=$PYTHONPATH:scenario_runner

export LEADERBOARD_ROOT=leaderboard
export CHALLENGE_TRACK_CODENAME=SENSORS
export PORT=2000
export TM_PORT=8000
export DEBUG_CHALLENGE=0
export REPETITIONS=3 # multiple evaluation runs
export RESUME=True

# TCP evaluation
export ROUTES=/home/psu/TCP/leaderboard/data/evaluation_routes/routes_lav_valid.xml
export TEAM_AGENT=leaderboard/team_code/tcp_agent.py
export TEAM_CONFIG=new.ckpt
export CHECKPOINT_ENDPOINT=results_TCP.json
export SCENARIOS=leaderboard/data/scenarios/all_towns_traffic_scenarios.json
export SAVE_PATH=data/results_TCP

python3 ${LEADERBOARD_ROOT}/leaderboard/leaderboard_evaluator.py \
--scenarios=${SCENARIOS}  \
--routes=${ROUTES} \
--repetitions=${REPETITIONS} \
--track=${CHALLENGE_TRACK_CODENAME} \
--checkpoint=${CHECKPOINT_ENDPOINT} \
--agent=${TEAM_AGENT} \
--agent-config=${TEAM_CONFIG} \
--debug=${DEBUG_CHALLENGE} \
--record=${RECORD_PATH} \
--resume=${RESUME} \
--port=${PORT} \
--trafficManagerPort=${TM_PORT}

`
Any help is greatly appreciated. Thanks!

Naive-Bayes, 2023-06-14:
> If you like, I can upload the model I trained in the whole dataset as the paper set. More discussion posts I publish here and screenshot here: [...]

I also meet this problem and I think this problem is not because the ckpt, but the scenario.json. Another issue for this problem you can reference here https://github.com/autonomousvision/transfuser/issues/53#issuecomment-1066847541

closed 2024-08-06 by penghao-wu