OpenDriveLab/TCP #15: Question on the frame rate used for data collection and control

Haichao-Zhang, 2022-10-28:
In the paper,  it is mentioned that "data samples are stored at 2HZ" (appendix A) and "future steps at 2HZ are predicted for both the trajectory branch and the control branch" (appendix B).
However, from the code, it seems that the control frequency is 20Hz:
https://github.com/OpenPerceptionX/TCP/blob/876cca2042009601e510ffe2b043c43b09ccbf35/leaderboard/leaderboard/leaderboard_evaluator.py#L90

Does this mean that we deploy the trained model with a higher framer ate? Or did I miss something?

Thanks!

WPH-commit, 2022-10-29:
Yes. When we deploy the model to interact with the environment, the model runs at 20Hz (the control is updated at 20Hz).

Haichao-Zhang, 2022-10-29:
Thanks for the confirmation!

closed 2022-10-29 by Haichao-Zhang