OpenDriveLab/TCP #16: Problem in Google Drive Dataset

Eyup-S, 2022-11-04:
I have tried multiple times trying to download dataset provided in google drive. It proceed for some time but it is cancelled due to "network error". Also i tried to download from baidu link which is mentioned in the issue (#12), but cannot reach the site. Could you share another link for dataset?

Kait0, 2023-02-28:
Experienced the same problem. The baidu mirror seems to work.

AOOOOOA, 2023-06-24:
> I have tried multiple times trying to download dataset provided in google drive. It proceed for some time but it is cancelled due to "network error". Also i tried to download from baidu link which is mentioned in the issue (#12), but cannot reach the site. Could you share another link for dataset?

me neither

AOOOOOA, 2023-06-24:
I solved this problem using the Firefox browser to download the Google Drive dataset. It may be because of some settings in google chrome( I previously used it) when downloading large files from Google Drive.  btw using Baidu Cloud to download the dataset is too slow...

xolovezari, 2023-12-19:
> I solved this problem using the Firefox browser to download the Google Drive dataset. It may be because of some settings in google chrome( I previously used it) when downloading large files from Google Drive. btw using Baidu Cloud to download the dataset is too slow...

ّ I tried your solution but it didn't work!
I really need to run the code, but the dataset can't be downloaded! Too sad :(

xolovezari, 2023-12-20:
> @xolovezari I will try to upload the dataset to huggingface. Are you able to access that?

I didn't try that before, but that would be great if you do me this favour! Thank you so much!

animikhaich, 2024-03-31:
> idn't try that before, but that would be great if you do me this favour! Thank you so much!
> please notice me when you upload that!

@penghao-wu, Can you please share the Hugginface version of the dataset? I have tried downloading from Google Drive, It fails after downloading ~100 GB with "network error". I am also unable to access the Baidu dataset since I am unable to create a Baidu account. 

Thanks!

penghao-wu, 2024-08-06:
> @penghao-wu, Can you please share the Hugginface version of the dataset? I have tried downloading from Google Drive, It fails after downloading ~100 GB with "network error". I am also unable to [...]

Sorry for the delay. You can now access the dataset in huggingface at https://huggingface.co/datasets/craigwu/tcp_carla_data