OpenDriveLab/TCP #22: some question about the network

crazyleeth, 2022-12-23:
thanks for your great work,which taught me a lot
however,I am a little confused about couple details about the network,looking for help.
1 You mentioned that you didnt use the last layer of the resnet34 which has 1000 output.but I find it in the code model.py
"self.join_traj = nn.Sequential(
							nn.Linear(128+1000, 512),
							nn.ReLU(inplace=True),
							nn.Linear(512, 512),
							nn.ReLU(inplace=True),
							nn.Linear(512, 256),
							nn.ReLU(inplace=True),
						)"
I am not sure about how this 1000 come from

2 it is about the attention map
"self.wp_att = nn.Sequential(
				nn.Linear(256+256, 256), 
				nn.ReLU(inplace=True),
				nn.Linear(256, 29*8),
				nn.Softmax(1)
			)"
I just start study the attention machanism  ，would you please show me what dose 29*8 represent？
looking forward to your reply : )

WPH-commit, 2022-12-23:
1. For the control branch, we do not use the last layer of the resnet34, but we use the last layer for the trajectory branch. You can modify the trajectory branch to discard the last layer to make it more consistent.
2. The 298 here is hard-coded to represent the size of the feature map (size 8*29) coming from the image backbone.

crazyleeth, 2022-12-24:
thank you for your reply！have a nice day~

closed 2022-12-24 by crazyleeth