OpenDriveLab/TCP #23: Request Test Model File

a1wj1, 2023-01-08:
Dear All,
I am a student engaged in automatic driving project. I'm currently running your code. Please provide us with your model ckpt file and results_ TCP.json file can be sent to us by Baidu Cloud if possible. Thank you.
thank you

WPH-commit, 2023-01-11:
Hi, you can download the ckpt file and the corresponding statistics through this link https://drive.google.com/file/d/1DaDCrSE6_SmpKa1bkrb3cOXxcnnHGQ8P/view?usp=sharing

Kait0, 2023-01-13:
Are these the proper pre-trained weights I asked for in https://github.com/OpenPerceptionX/TCP/issues/11 ?

WPH-commit, 2023-01-13:
> Are these the proper pre-trained weights I asked for in #11 ?

These are the pre-trained weights for the ablations experiments in the paper. We do not plan to release weights for leaderboard submission as it is an ensemble of multiple models.

tenderzada, 2024-11-23:
> Hi, you can download the ckpt file and the corresponding statistics through this link https://drive.google.com/file/d/1DaDCrSE6_SmpKa1bkrb3cOXxcnnHGQ8P/view?usp=sharing

Dear authors,

Thank you for sharing this great framework! I am currently learning CARLA and exploring LLM-based end-to-end autonomous driving. I noticed the checkpoint link has expired—could you kindly re-share it along with any relevant results?

Thank you for your help!

Guji0302, 2025-02-04:
> Hi, you can download the ckpt file and the corresponding statistics through this link https://drive.google.com/file/d/1DaDCrSE6_SmpKa1bkrb3cOXxcnnHGQ8P/view?usp=sharing
Hello, the link you shared is no longer valid, could you please send it again

Guji0302, 2025-02-04:
> Dear All, I am a student engaged in automatic driving project. I'm currently running your code. Please provide us with your model ckpt file and results_ TCP.json file can be sent to us by Baidu Cloud if possible. Thank you. thank you
Hello, the link you shared is no longer valid, could you please send it again

TianYifan-Vincent, 2025-03-06:
@penghao-wu Could you please send it again?