OpenDriveLab/TCP #24: Visualizing waypoints

macandro96, 2023-02-13:
While running the evaluation, I noticed that the `tcp_agent.py` dumps the network predictions in a json file. Is is possible to project these waypoint coordinates on the image? 

Thanks!

WPH-commit, 2023-02-13:
Yes, the saved waypoints coordinate is [right, forward]. You can project them to the image using intrinsic and extrinsic matrices.
For the intrinsic matrix, the fov is 100 degrees, Cu = 900/2=450, Cv = 256/2 = 128.
For the extrinsic matrix, the coordinate of the camera is ```'x': -1.5, 'y': 0.0, 'z':2.0,``` with respect to the vehicle.

macandro96, 2023-02-13:
Thank you so much. Just to confirm, does this logic look okay?
```
x_c, y_c, z_c = -1.5, 0.0, 2.0  # camera centers
C_u, C_v = 900 / 2, 256 / 2
F_x = 900 /(2 * np.tan(100 * np.pi / 360))
F_y = 256 /(2 * np.tan(100 * np.pi / 360))

# calculating extrinsic matrix
E = np.zeros((3, 4))

# Using E = [R | t]
# R is Yaw * Pitch * Roll (all are 0)
E[:3, :3] = np.eye(3)  # R = Identity
E[:, 3] = -np.array([x_c, y_c, z_c])  # t = - R C

# Intrinsic camera matrix
K = np.array([
    [F_x, 0, C_u],
    [0, F_y, C_v],
    [0, 0, 1]
])

# waypoint predictions
out = {
    'wp1': (-0.26721786, -2.74913808),
    'wp2': (-1.09084616, -5.1806664),
    'wp3': (-2.53841191, -7.36515022),
    'wp4': (-4.54262737, -8.98083773)
}
# stacking waypoint predictions
wp = np.zeros((4, 4))
wp[0:2, 0] = np.array(out['wp1'])
wp[0:2, 1] = np.array(out['wp2'])
wp[0:2, 2] = np.array(out['wp3'])
wp[0:2, 3] = np.array(out['wp4'])
# z coordinate is set to zero

# setting homogenous coordinate to 1
wp[-1, :] = np.ones(4)

img_coords = E @ K @ wp
```

WPH-commit, 2023-02-15:
First, make sure the waypoint coordinate is [forward, left]. Then the extrinsic matrix should be [[0,-1,0,0],[0,0,-1,2],[1,0,0,1.5],[0,0,0,1]]

macandro96, 2023-02-18:
Got it. Thank you so much.

closed 2023-02-18 by macandro96

dinuransika, 2023-06-21:
Can someone show how to plot this image coordinates on the front_img

azamshoaib, 2023-09-06:
@macandro96 @penghao-wu 

Can you please provide the visualization code.

626960883, 2025-05-09:
Hello, I would like to consult about the problem of generating visualization of path points in the graph. First of all, is the json file in the meta in the results_TCP folder the path point? How to map the path point on the corresponding RGB image? Thank you！

Below is the file content of the corresponding meta json：
0001.json
"speed":1.88073876472572e-08steer":0.0010592935632724226throttle":0.0,brake":1.0,Wp 4":0.0005435124039649963-0.006604880094528198
Wp 3":5.455315113067627e-050.010030608624219894
Wp_2":「0.00195121765136718750.004142649471759796
wp 1”:[0.008278928697109222-0.006492529064416885
'aim":5.455315113067627e-050.010030608624219894
target":0.0922911092638969451.9973030090332
'desired speed":0.020111518135915198angle":2.0034623266088056，angle last":0.11003249831759156"angle target":0.0011299131341572508"angle final":0.0011299131341572508"delta":0.02011149932852755，
'agent":"traj",'steer ctrl":-0.07172626256942749"steer trai":0.0010592935632724226