OpenDriveLab/TCP #25: Excuse me,  there are some operational problems about "python TCP/train.py --gpus"

luoyexx, 2023-02-17:
Excuse me,  there are some operational problems about "python TCP/train.py --gpus"

python TCP/train.py --gpus 1
Traceback (most recent call last):
  File "TCP/train.py", line 171, in <module>
    train_set = CARLA_Data(root=config.root_dir_all, data_folders=config.train_data, img_aug = config.img_aug)
  File "/home/zjy/TCP/TCP/TCP/data.py", line 49, in __init__
    data = np.load(os.path.join(sub_root, "packed_data.npy"), allow_pickle=True).item()
  File "/home/zjy/anaconda3/envs/TCP/lib/python3.7/site-packages/numpy/lib/npyio.py", line 416, in load
    fid = stack.enter_context(open(os_fspath(file), "rb"))
FileNotFoundError: [Errno 2] No such file or directory: 'Tcp/tcp_carla_data/town01/packed_data.npy'

WPH-commit, 2023-02-18:
Hi, you should set the root_dir_all variable in config.py to the actual directory of your data.

luoyexx, 2023-02-18:
So sorry,  I have set the root_dir_all variable for dataset in config.py.  I use the Baidu Cloud dataset provided by you. I found that there is no packed_ data.npy in this dataset.

WPH-commit, 2023-02-19:
Sorry about this problem. Those files might get ignored for the Baidu Cloud zip file. You can use ```tools/gen_data.py``` to generate packed_data.npy files.

luoyexx, 2023-02-22:
> Sorry about this problem. Those files might get ignored for the Baidu Cloud zip file. You can use `tools/gen_data.py` to generate packed_data.npy files.

Thank. And another question.
(TCP) zjy@wp:~/TCP$ python TCP/train.py
0
0
Traceback (most recent call last):
  File "TCP/train.py", line 177, in <module>
    dataloader_val = DataLoader(val_set, batch_size=args.batch_size, shuffle=True, num_workers=8)
  File "/home/zjy/anaconda3/envs/TCP/lib/python3.7/site-packages/torch/utils/data/dataloader.py", line 268, in __init__
    sampler = RandomSampler(dataset, generator=generator)
  File "/home/zjy/anaconda3/envs/TCP/lib/python3.7/site-packages/torch/utils/data/sampler.py", line 103, in __init__
    "value, but got num_samples={}".format(self.num_samples))

ValueError: num_samples should be a positive integer value, but got num_samples=0

WPH-commit, 2023-02-24:
Could you please check the content of the ```packed_data.npy``` to see if it is correctly generated?

luoyexx, 2023-03-08:
Exception ignored in: <function LeaderboardEvaluator.__del__ at 0x7f81f5933200>
Traceback (most recent call last):
  File "leaderboard/leaderboard/leaderboard_evaluator.py", line 150, in __del__
    self._cleanup()
  File "leaderboard/leaderboard/leaderboard_evaluator.py", line 162, in _cleanup
    if self.manager and self.manager.get_running_status() \
AttributeError: 'LeaderboardEvaluator' object has no attribute 'manager'
Traceback (most recent call last):
  File "leaderboard/leaderboard/leaderboard_evaluator.py", line 529, in <module>
    main()
  File "leaderboard/leaderboard/leaderboard_evaluator.py", line 525, in main
    del leaderboard_evaluator
UnboundLocalError: local variable 'leaderboard_evaluator' referenced before assignment

WPH-commit, 2023-03-14:
Could you please provide all the error information? Usually, this del leaderboard_evaluator error happens after some other exceptions or errors.

SExpert12, 2024-06-27:
Are you able to solve this error? I run 

sh leaderboard/scripts/data_collection.sh

and I get this error:

init statistics_manager
begin
Traceback (most recent call last):
  File "leaderboard/leaderboard/leaderboard_evaluator.py", line 519, in main
    leaderboard_evaluator = LeaderboardEvaluator(arguments, statistics_manager)
  File "leaderboard/leaderboard/leaderboard_evaluator.py", line 123, in __init__
    self.module_agent = importlib.import_module(module_name)
  File "/home/ryzen/miniconda3/envs/TCPagent/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 728, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/home/ryzen/TCP/leaderboard/team_code/roach_ap_agent.py", line 17, in <module>
    from roach.criteria import run_stop_sign
ModuleNotFoundError: No module named 'roach'
Exception ignored in: <function LeaderboardEvaluator.__del__ at 0x7f1d54cd43b0>
Traceback (most recent call last):
  File "leaderboard/leaderboard/leaderboard_evaluator.py", line 150, in __del__
    self._cleanup()
  File "leaderboard/leaderboard/leaderboard_evaluator.py", line 162, in _cleanup
    if self.manager and self.manager.get_running_status() \
AttributeError: 'LeaderboardEvaluator' object has no attribute 'manager'
Traceback (most recent call last):
  File "leaderboard/leaderboard/leaderboard_evaluator.py", line 529, in <module>
    main()
  File "leaderboard/leaderboard/leaderboard_evaluator.py", line 525, in main
    del leaderboard_evaluator
UnboundLocalError: local variable 'leaderboard_evaluator' referenced before assignment

How to resolve this?