OpenDriveLab/TCP #26: Set the dataset path to``TCP/config.py``

hting147, 2023-02-23:
Sorry for the qusetion, I have downloaded all the datasets from Baidu. But I do not know how to set the dataset path in ``TCP/config.py``.
I am using window os. Thank you

WPH-commit, 2023-02-24:
Hi, could you please try to unzip the ```tcp_carla_data_split.zip```? And other files should be automatically unzipped as well. Then you can set the dataset path to the path of the unzipped folder.

hting147, 2023-02-24:
yes, I have unzipped all the dataset folders, but I could not find a way to set the path to the unzipped folder. Is it use the notepad to set the path to the dataset folder, or is there another way? thank you for  the response

WPH-commit, 2023-02-24:
Sorry for the possible confusion. "Setting the path" means setting the ```root_dir_all``` variable in the ```config.py``` file.

hting147, 2023-02-24:
So do I need to add anything or a command in the config.py file? sorry for my stupid question, it is because this is my first time using Carla. Thank you for your patience

WPH-commit, 2023-02-24:
You just need to change the ```root_dir_all = "tcp_carla_data"``` to your actual unzipped data directory.

WPH-commit, 2023-02-28:
Hi, you should replace ```CARLA_ROOT``` with the actual path to your Carla directory

hting147, 2023-03-01:
ok, I have opened the carla.exe, but i do not have any response after writing any command such as 
sh leaderboard/scripts/data_collection.sh. Is it need to open another cmd screen to write any command? thank you

WPH-commit, 2023-03-14:
could you provide the terminal output after you run sh leaderboard/scripts/data_collection.sh?