OpenDriveLab/TCP #39: code

a1wj1, 2023-07-06:
This code doesn't match the formulas in the paper. Could you please help explain,?

[image]
[image]
[image]

a1wj1, 2023-07-06:
In j_ctrl = self.join_ctrl(torch.cat([feature_emb, measurement_feature], 1))，why not h_ctl but measurement_feature

WPH-commit, 2023-07-09:
This is the initial step so the measurement feature is used instead of h_ctrl.
[image]

closed 2023-07-12 by WPH-commit