OpenDriveLab/TCP #45: filter_data

a1wj1, 2023-07-22:
When running tools/filter_data.py, the path cannot be found：
FileNotFoundError: [Errno 2] No such file or directory: 'routes_town05_long.json'

May I ask if this file will be generated when collecting data? I don't seem to have found the relevant generation code?

WPH-commit, 2023-07-24:
It should correspond to the file specified by CHECKPOINT_ENDPOINT in the data_collection script.

noah-curran, 2023-07-26:
@WPH-commit could you please clarify which file paths should be set to `result_path` and `data_path` in filter_data.py? Please provide an example. I have successfully generated data using `routes_town01.xml` with roach collection.

BTW, as a related note, it would be greatly appreciated if the documentation could be updated with more specific steps of what to do with these parameters during all steps. As the documentation current is, it has been a lot of trial and error to figure out what is to be filled where. I'd like to reproduce your results from start to finish.