OpenDriveLab/TCP #51: Question about data collection

xanhug, 2023-09-12:
Thanks for your amazing work!
when I try to generate dataset there is a error
```
(TCP) hxa@hxa-Nitro-AN515-58:~/TCP$ sh leaderboard/scripts/data_collection.sh
init statistics_manager
begin

========= Preparing RouteScenario_0 (repetition 0) =========
> Setting up the agent
> Loading the world
Town01
load
load
load

The scenario could not be loaded:
> set_global_plan() missing 1 required positional argument: 'wp_route'

Traceback (most recent call last):
  File "leaderboard/leaderboard/leaderboard_evaluator.py", line 352, in _load_and_run_scenario
    scenario = RouteScenario(world=self.world, config=config, debug_mode=args.debug)
  File "/home/hxa/InterFuser/leaderboard/leaderboard/scenarios/route_scenario.py", line 189, in __init__
    self._update_route(world, config, debug_mode>0)
  File "/home/hxa/InterFuser/leaderboard/leaderboard/scenarios/route_scenario.py", line 229, in _update_route
    config.agent.set_global_plan(gps_route, self.route)
TypeError: set_global_plan() missing 1 required positional argument: 'wp_route'
> Registering the route statistics
Exception ignored in: <function RouteScenario.__del__ at 0x7f15ef125950>
Traceback (most recent call last):
  File "/home/hxa/InterFuser/leaderboard/leaderboard/scenarios/route_scenario.py", line 572, in __del__
    self.remove_all_actors()
  File "/home/hxa/InterFuser/scenario_runner/srunner/scenarios/basic_scenario.py", line 202, in remove_all_actors
    for i, _ in enumerate(self.other_actors):
AttributeError: 'RouteScenario' object has no attribute 'other_actors'
```
Could you please tell me how to deal with it?

By the way, the trainning and evaluation are both running properly.

xanhug, 2023-09-12:
And here is my data_collection file
```
#!/bin/bash
export CARLA_ROOT=/home/hxa/TCP/carla
export CARLA_SERVER=${CARLA_ROOT}/CarlaUE4.sh
export PYTHONPATH=$PYTHONPATH:${CARLA_ROOT}/PythonAPI
export PYTHONPATH=$PYTHONPATH:${CARLA_ROOT}/PythonAPI/carla
export PYTHONPATH=$PYTHONPATH:$CARLA_ROOT/PythonAPI/carla/dist/carla-0.9.10-py3.7-linux-x86_64.egg
export PYTHONPATH=$PYTHONPATH:leaderboard
export PYTHONPATH=$PYTHONPATH:leaderboard/team_code
export PYTHONPATH=$PYTHONPATH:scenario_runner

export LEADERBOARD_ROOT=leaderboard
export CHALLENGE_TRACK_CODENAME=SENSORS
export PORT=2000
export TM_PORT=8000
export DEBUG_CHALLENGE=0
export REPETITIONS=1 # multiple evaluation runs
export RESUME=True
export DATA_COLLECTION=True

# Roach data collection
export ROUTES=leaderboard/data/TCP_training_routes/routes_town01.xml
export TEAM_AGENT=team_code/roach_ap_agent.py
export TEAM_CONFIG=roach/config/config_agent.yaml
export CHECKPOINT_ENDPOINT=data_collect_town01_results.json
export SCENARIOS=leaderboard/data/scenarios/all_towns_traffic_scenarios.json
export SAVE_PATH=data/data_collect_town01_results/

python3 ${LEADERBOARD_ROOT}/leaderboard/leaderboard_evaluator.py \
--scenarios=${SCENARIOS}  \
--routes=${ROUTES} \
--repetitions=${REPETITIONS} \
--track=${CHALLENGE_TRACK_CODENAME} \
--checkpoint=${CHECKPOINT_ENDPOINT} \
--agent=${TEAM_AGENT} \
--agent-config=${TEAM_CONFIG} \
--debug=${DEBUG_CHALLENGE} \
--record=${RECORD_PATH} \
--resume=${RESUME} \
--port=${PORT} \
--trafficManagerPort=${TM_PORT}

```

xanhug, 2023-09-12:
I have found the reason.
my environment variables was not clear and the program make a mistake to locate at a File with the same name for other projects.
now the problem have been solved，sorry for bother.

closed 2023-09-12 by xanhug

HXTYI, 2024-07-13:
I wanna ask if your data_collection file right.