OpenDriveLab/TCP #57: Could you please provide the script to generate town06_addition?

xanhug, 2023-11-01:
Thanks for your amazing work!
when I try to generate dataset, I found there is no script of town06_addition compared to your provided dataset.
could you please the script to generate town06_addition?
[image]
[image]

penghao-wu, 2023-11-01:
Hi, I have uploaded the missing xml file. Thank you for pointing this out. And we use tools/generate_random_routes.py to generate these routes.

xanhug, 2023-11-02:
> Hi, I have uploaded the missing xml file. Thank you for pointing this out. And we use tools/generate_random_routes.py to generate these routes.

Thank you so much!

closed 2023-11-02 by xanhug