OpenDriveLab/TCP #60: How to integrate a TCP agent into the new version of Carla(0.9.14)?

Chionanthus, 2024-01-03:
Hello, I am currently planning to use TCP as a tested autonomous driving agent for further research. I have trained the model, and when running 'run_evaluation.sh,' I encounter an error:
'''File "/home/dell/TYQ/TCP/leaderboard/leaderboard/autoagents/agent_wrapper.py", line 21, in <module>
    from leaderboard.autoagents.autonomous_agent import Track
  File "/home/dell/TYQ/TCP/leaderboard/leaderboard/autoagents/autonomous_agent.py", line 17, in <module>
    from leaderboard.utils.route_manipulation import downsample_route
  File "/home/dell/TYQ/TCP/leaderboard/leaderboard/utils/route_manipulation.py", line 17, in <module>
    from agents.navigation.global_route_planner_dao import GlobalRoutePlannerDAO
'''
Due to my intention to conduct research using Carla's Town12 map, the Carla version I am using is 0.9.14. I have identified that the issue is due to the absence of the file 'global_route_planner_dao' in the directory 'CARLA_ROOT/PythonAPI/carla/agent/navigation' (while 'global_route_planner' is present), causing the execution to fail.

So I want to know how to integrate TCP into a newer version of Carla. I look forward to your prompt reply. Thank you.

Chionanthus, 2024-01-03:
我将 Carla 0.9.10.1 中的这个文件复制到 0.9.14 对应的文件夹里，这个问题解决了，但是又有新的报错。
目前是想把 TCP 作为一个独立的 agent 接入新版本 Carla(0.9.14) 即可，原本 LeaderBoard 的部分可以不用理。请问作者知道该怎么调整吗？

penghao-wu, 2024-01-03:
I think you should only export the team_code folder to the new version leaderboard and make modifications accordingly.

Chionanthus, 2024-01-03:
My plan is independent of the Carla LeaderBoard. 
In my research, I have set up new test scenarios in Town12. I intend to extract TCP as a standalone agent, similar to Carla's automatic_control.py, specifying start and end points to evaluate the excellent performance of TCP. 
However, this work is still in the early stages, and I am unsure how to extract TCP as a standalone autonomous driving agent in Carla 0.9.14. I will continue to experiment, and any progress will be mentioned in this ISSUE. If you have any insights, your assistance would be appreciated.

Chionanthus, 2024-01-03:
> I think you should only export the team_code folder to the new version leaderboard and make modifications accordingly.

btw, I would like to know if all the code related to the actual use of the autonomous driving agent is only in the 'team_code' folder along with the pre-trained models. 
Are the remaining files only related to the leaderboard, which I don't need to pay excessive attention to?

penghao-wu, 2024-01-03:
> btw, I would like to know if all the code related to the actual use of the autonomous driving agent is only in the 'team_code' folder along with the pre-trained models. Are the remaining files only [...]

Yes. But you might also need to check whether some APIs of the new leaderboard used in our agent have changed.

TahaCarla, 2024-05-16:
> btw, I would like to know if all the code related to the actual use of the autonomous driving agent is only in the 'team_code' folder along with the pre-trained models. Are the remaining files only [...]

Hello I am also having the same issue if you find a solution for it, please let me know.

HHUBRO, 2024-08-30:
你好，请问问题解决了吗？我也遇到同样的问题，现在我用roach采数据，只能运行一段场景，下一段场景就会报错，