OpenDriveLab/TCP #61: AssertionError: Expected image to have shape (height, width, [channels]), got shape ().

donkehuang, 2024-01-04:
When i run the training progress with `python TCP/train.py`, it works util the "Saving latest checkpoint". 
When Saving latest checkpoint, it occurs the error that 
`AssertionError: Caught AssertionError in DataLoader worker process 7.
Original Traceback (most recent call last):
  File "/home/donke/anaconda3/envs/TCP/lib/python3.7/site-packages/torch/utils/data/_utils/worker.py", line 287, in _worker_loop
    data = fetcher.fetch(index)
  File "/home/donke/anaconda3/envs/TCP/lib/python3.7/site-packages/torch/utils/data/_utils/fetch.py", line 49, in fetch
    data = [self.dataset[idx] for idx in possibly_batched_index]
  File "/home/donke/anaconda3/envs/TCP/lib/python3.7/site-packages/torch/utils/data/_utils/fetch.py", line 49, in <listcomp>
    data = [self.dataset[idx] for idx in possibly_batched_index]
  File "/home/donke/TCP/TCP/data.py", line 90, in __getitem__
    Image.open(self.root+self.front_img[index][0]))))
  File "/home/donke/anaconda3/envs/TCP/lib/python3.7/site-packages/imgaug/augmenters/meta.py", line 769, in augment_image
    "got shape %s." % (image.shape,))
AssertionError: Expected image to have shape (height, width, [channels]), got shape ().
`
what should i do, thanks

qqcole2000, 2024-05-11:
遇到同样的问题，请问解决了吗

qqcole2000, 2024-05-11:
Hello, I have encountered the same problem,have you resolved it?

donkehuang, 2024-05-12:
> Hello, I have encountered the same problem,have you resolved it?

I have solve this problem which is caused by an image with incorrect format. To solve this bug, you can skip the image with incorrect format or just delete the image in dataset. The specific index of the image is hard for me to remember, but i hope the code for skipping the image may  help you.
In data.py:88,correct the code like this:
`		if self.img_aug:
			try:
				temp = np.array(Image.open(self.root+self.front_img[index][0]))
				data['front_img'] = self._im_transform(augmenter(self._batch_read_number).augment_image(temp))
			except:
				temp = np.array(Image.open(self.root+self.front_img[0][0]))
				data['front_img'] = self._im_transform(augmenter(self._batch_read_number).augment_image(temp))
		else:
			try:
				temp = np.array(Image.open(self.root+self.front_img[index][0]))
				data['front_img'] = self._im_transform(temp)		
			except:
				temp = np.array(Image.open(self.root+self.front_img[0][0]))
				data['front_img'] = self._im_transform(temp)		`

closed 2024-05-12 by donkehuang

qqcole2000, 2024-05-13:
Thanks for your reply, it helps me a lot.