OpenDriveLab/TCP #62: The question about setting TEAM_CONFIG path in Evaluation

ml6362662, 2024-05-24:
Hello, I'm a beginner carla autopilot student, mentioned in the Evaluation section of your README.md file, you need to manually set the path of model ckpt at the TEAM_CONFIG in the leaderboard/scripts/run_evaluation.sh file, but I don't know which model ckpt is this model configuration file, or how to get model ckpt. I would like to ask you about this TEAM_ How CONFIG should be set up, thank you very much.
[image]
[image]

HXTYI, 2024-06-24:
Have you solved this problem? I have encountered this too.

zdy1013, 2024-08-05:
This is the path to the ckpt file you trained yourself to get.

closed 2024-08-06 by penghao-wu

1585596474, 2024-10-12:
> This is the path to the ckpt file you trained yourself to get.

Hi, I trained the network, but I don't know the directory of the ckpt file. Can you tell me the default directory?