OpenDriveLab/TCP #65: 'LeaderboardEvaluator' object has no attribute 'manager'

SExpert12, 2024-06-27:
Hi, 
I got this error when I run sh leaderboard/scripts/data_collection.sh:
init statistics_manager
begin
Traceback (most recent call last):
  File "leaderboard/leaderboard/leaderboard_evaluator.py", line 519, in main
    leaderboard_evaluator = LeaderboardEvaluator(arguments, statistics_manager)
  File "leaderboard/leaderboard/leaderboard_evaluator.py", line 123, in __init__
    self.module_agent = importlib.import_module(module_name)
  File "/home/ryzen/miniconda3/envs/TCPagent/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 728, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/home/ryzen/TCP/leaderboard/team_code/roach_ap_agent.py", line 17, in <module>
    from roach.criteria import run_stop_sign
ModuleNotFoundError: No module named 'roach'
Exception ignored in: <function LeaderboardEvaluator.__del__ at 0x7f1d54cd43b0>
Traceback (most recent call last):
  File "leaderboard/leaderboard/leaderboard_evaluator.py", line 150, in __del__
    self._cleanup()
  File "leaderboard/leaderboard/leaderboard_evaluator.py", line 162, in _cleanup
    if self.manager and self.manager.get_running_status() \
AttributeError: 'LeaderboardEvaluator' object has no attribute 'manager'
Traceback (most recent call last):
  File "leaderboard/leaderboard/leaderboard_evaluator.py", line 529, in <module>
    main()
  File "leaderboard/leaderboard/leaderboard_evaluator.py", line 525, in main
    del leaderboard_evaluator
UnboundLocalError: local variable 'leaderboard_evaluator' referenced before assignment

And although this is correct, I comment this line and then I got this error:
init statistics_manager
begin
Traceback (most recent call last):
  File "leaderboard/leaderboard/leaderboard_evaluator.py", line 519, in main
    leaderboard_evaluator = LeaderboardEvaluator(arguments, statistics_manager)
  File "leaderboard/leaderboard/leaderboard_evaluator.py", line 123, in __init__
    self.module_agent = importlib.import_module(module_name)
  File "/home/ryzen/miniconda3/envs/TCPagent/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 728, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/home/ryzen/TCP/leaderboard/team_code/roach_ap_agent.py", line 17, in <module>
    from roach.criteria import run_stop_sign
ModuleNotFoundError: No module named 'roach'
Exception ignored in: <function LeaderboardEvaluator.__del__ at 0x7f4ea2923b90>
Traceback (most recent call last):
  File "leaderboard/leaderboard/leaderboard_evaluator.py", line 150, in __del__
    self._cleanup()
  File "leaderboard/leaderboard/leaderboard_evaluator.py", line 162, in _cleanup
    if self.manager and self.manager.get_running_status() \
AttributeError: 'LeaderboardEvaluator' object has no attribute 'manager'

How to resolve this now?

HXTYI, 2024-07-11:
I meet the same problem, have you solved this ?

SExpert12, 2024-07-12:
Not yet.

HXTYI, 2024-07-12:
when you open CarlaUE4.sh, the problem seems be sovled.

SExpert12, 2024-07-12:
Okay. Let me check.

HXTYI, 2024-07-12:
I am also reproducing this project recently. Maybe we can have more communication and reproduce it together. My email is yeagerXu@gmail.com.

HXTYI, 2024-07-12:
I also met the problem that ModuleNotFoundError: No module named 'roach'. I added in the path, but its not work.

SExpert12, 2024-07-13:
Can you tell me what is your project? I mean basic configuration about your project.

penghao-wu, 2024-08-06:
> Hi, I got this error when I run sh leaderboard/scripts/data_collection.sh: init statistics_manager begin Traceback (most recent call last): File "leaderboard/leaderboard/leaderboard_evaluator.py", line 519, in main leaderboard_evaluator = LeaderboardEvaluator(arguments, statistics_manager) File "leaderboard/leaderboard/leaderboard_evaluator.py", line 123, in **init** self.module_agent = importlib.import_module(module_name) File "/home/ryzen/miniconda3/envs/TCPagent/lib/python3.7/importlib/**init**.py", line 127, in import_module return _bootstrap._gcd_import(name[level:], package, level) File "", line 1006, in _gcd_import File "", line 983, in _find_and_load File "", line 967, in _find_and_load_unlocked File "", line 677, in _load_unlocked File "", line 728, in exec_module File "", line 219, in _call_with_frames_removed File "/home/ryzen/TCP/leaderboard/team_code/roach_ap_agent.py", line 17, in from roach.criteria import run_stop_sign ModuleNotFoundError: No module named 'roach' Exception ignored in: <function LeaderboardEvaluator.**del** at 0x7f1d54cd43b0> Traceback (most recent call last): File "leaderboard/leaderboard/leaderboard_evaluator.py", line 150, in **del** self._cleanup() File "leaderboard/leaderboard/leaderboard_evaluator.py", line 162, in _cleanup if self.manager and self.manager.get_running_status() AttributeError: 'LeaderboardEvaluator' object has no attribute 'manager' Traceback (most recent call last): File "leaderboard/leaderboard/leaderboard_evaluator.py", line 529, in main() File "leaderboard/leaderboard/leaderboard_evaluator.py", line 525, in main del leaderboard_evaluator UnboundLocalError: local variable 'leaderboard_evaluator' referenced before assignment
> 
> And although this is correct, I comment this line and then I got this error: init statistics_manager begin Traceback (most recent call last): File "leaderboard/leaderboard/leaderboard_evaluator.py", line 519, in main leaderboard_evaluator = LeaderboardEvaluator(arguments, statistics_manager) File "leaderboard/leaderboard/leaderboard_evaluator.py", line 123, in **init** self.module_agent = importlib.import_module(module_name) File "/home/ryzen/miniconda3/envs/TCPagent/lib/python3.7/importlib/**init**.py", line 127, in import_module return _bootstrap._gcd_import(name[level:], package, level) File "", line 1006, in _gcd_import File "", line 983, in _find_and_load File "", line 967, in _find_and_load_unlocked File "", line 677, in _load_unlocked File "", line 728, in exec_module File "", line 219, in _call_with_frames_removed File "/home/ryzen/TCP/leaderboard/team_code/roach_ap_agent.py", line 17, in from roach.criteria import run_stop_sign ModuleNotFoundError: No module named 'roach' Exception ignored in: <function LeaderboardEvaluator.**del** at 0x7f4ea2923b90> Traceback (most recent call last): File "leaderboard/leaderboard/leaderboard_evaluator.py", line 150, in **del** self._cleanup() File "leaderboard/leaderboard/leaderboard_evaluator.py", line 162, in _cleanup if self.manager and self.manager.get_running_status() AttributeError: 'LeaderboardEvaluator' object has no attribute 'manager'
> 
> How to resolve this now?

Have you added the TCP path to PYTHONPATH like this `export PYTHONPATH=$PYTHONPATH:PATH_TO_TCP`?