OpenDriveLab/TCP #66: Traceback (most recent call last):   File "/home/cybersec/TCP-main/TCP/train.py", line 15, in <module>     from TCP.model import TCP ModuleNotFoundError: No module named 'TCP'

Mr-ChenSH, 2024-07-31:
Traceback (most recent call last):
  File "/home/cybersec/TCP-main/TCP/train.py", line 15, in <module>
    from TCP.model import TCP
ModuleNotFoundError: No module named **'TCP'***

Encountering this issue while running train.py

penghao-wu, 2024-08-06:
Have you added the TCP path to PYTHONPATH like this `export PYTHONPATH=$PYTHONPATH:PATH_TO_TCP`?

Mr-ChenSH, 2024-08-11:
yes.
export CARLA_ROOT=/home/cybersec/Work/Docker/carla/share/PythonAPI/carla

Mr-ChenSH, 2024-08-11:
The error is:
Traceback (most recent call last):
  File "/home/cybersec/TCP-main/leaderboard/leaderboard/leaderboard_evaluator.py", line 34, in <module>
    from leaderboard.scenarios.scenario_manager import ScenarioManager
  File "/home/cybersec/TCP-main/leaderboard/leaderboard/scenarios/scenario_manager.py", line 25, in <module>
    from leaderboard.autoagents.agent_wrapper import AgentWrapper, AgentError
  File "/home/cybersec/TCP-main/leaderboard/leaderboard/autoagents/agent_wrapper.py", line 21, in <module>
    from leaderboard.autoagents.autonomous_agent import Track
  File "/home/cybersec/TCP-main/leaderboard/leaderboard/autoagents/autonomous_agent.py", line 17, in <module>
    from leaderboard.utils.route_manipulation import downsample_route
  File "/home/cybersec/TCP-main/leaderboard/leaderboard/utils/route_manipulation.py", line 22, in <module>
    from agents.navigation.global_route_planner import GlobalRoutePlanner
ModuleNotFoundError: No module named 'agents.navigation'

Mr-ChenSH, 2024-08-14:
How to change the perspective of the car during verification

closed 2024-08-25 by Mr-ChenSH