OpenDriveLab/TCP #67: Reproducing the results of the questioning

zdy1013, 2024-08-01:
I'm glad you published such a great work, why are the results I get so different from yours when I use 2 RTX 3090s and train with 60 epochs using the dataset you provided and evaluate it? Here are the results I reproduced:
Avg. driving score: 52.005
Avg. route completion: 85.111
Avg. infraction penalty: 0.647
Collisions with pedestrians: 0.000
Collisions with vehicles: 0.270
Collisions with layout: 0.097
Red lights infractions: 0.070
Stop sign infractions: 0.238
Off-road infractions: 0.198
Route deviations: 0.000
Route timeouts: 0.094
Agent blocked: 0.298

penghao-wu, 2024-08-06:
> I'm glad you published such a great work, why are the results I get so different from yours when I use 2 RTX 3090s and train with 60 epochs using the dataset you provided and evaluate it? Here are [...]

I suppose you are evaluating the model in 48 routes where our model has 57.01 driving score as reported in the paper. I think the difference is reasonable considering the variance in evaluation and training.

zdy1013, 2024-08-06:
> I suppose you are evaluating the model in 48 routes where our model has 57.01 driving score as reported in the paper. I think the difference is reasonable considering the variance in evaluation and [...]

Thank you for your reply. May I ask how to operate if I want to reproduce your score of 75.14?
[image]

penghao-wu, 2024-08-06:
We use all 420K data for training. And the ensemble of the TCP and TCP-SB models is used. Some details can be found in the paper.
[image]

zdy1013, 2024-08-06:
> We use all 420K data for training. And the ensemble of the TCP and TCP-SB models is used. Some details can be found in the paper. [...]

I would like to know, how do you integrate?What's more, is the data set you provided for 4 towns?

penghao-wu, 2024-08-06:
The dataset file should contain all 8 towns' data. For the ensemble strategy, please refer to the last sentence in the image.

HXTYI, 2024-08-08:
Hello author, I recently reproduced this project, but the test score is similar to the above figure, between 50-60. What do you mean by 420K data? Is this 115G tcp_carla_data?
If I want the results in the figure, do I also need to "If it is a trajectory specialization case, we set α=0.5, if it is a control specialization case, we take the maximum value of the brake control instead of the average value"
Hope u can reply !

HXTYI, 2024-08-08:
@penghao-wu 
Hello author, I recently reproduced this project, but the test score is similar to the above figure, between 50-60. What do you mean by 420K data? Is this 115G tcp_carla_data?
If I want the results in the figure, do I also need to "If it is a trajectory specialization case, we set α=0.5, if it is a control specialization case, we take the maximum value of the brake control instead of the average value"
Hope u can reply !

penghao-wu, 2024-08-08:
> @penghao-wu Hello author, I recently reproduced this project, but the test score is similar to the above figure, between 50-60. What do you mean by 420K data? Is this 115G tcp_carla_data? If I want [...]

Yes, the 420K data includes all 8 towns' data in the data zip.
Yes.

Mr-ChenSH, 2024-08-12:
@penghao-wu  OK，I konw it.

closed 2024-08-12 by penghao-wu