OpenDriveLab/TCP #68: Traceback (most recent call last):   File "/home/cybersec/TCP-main/leaderboard/leaderboard/leaderboard_evaluator.py", line 34, in <module>     from leaderboard.scenarios.scenario_manager import ScenarioManager   File "/home/cybersec/TCP-main/leaderboard/leaderboard/scenarios/scenario_manager.py", line 25, in <module>     from leaderboard.autoagents.agent_wrapper import AgentWrapper, AgentError   File "/home/cybersec/TCP-main/leaderboard/leaderboard/autoagents/agent_wrapper.py", line 21, in <module>     from leaderboard.autoagents.autonomous_agent import Track   File "/home/cybersec/TCP-main/leaderboard/leaderboard/autoagents/autonomous_agent.py", line 17, in <module>     from leaderboard.utils.route_manipulation import downsample_route   File "/home/cybersec/TCP-main/leaderboard/leaderboard/utils/route_manipulation.py", line 16, in <module>     from agents.navigation.global_route_planner import GlobalRoutePlanner ModuleNotFoundError: No module named 'agents.navigation'

Mr-ChenSH, 2024-08-11:
run  sh leaderboard/scripts/data_collection.sh encounter this problem

penghao-wu, 2024-08-11:
Could you please modify the issue title to make it concise? For you problem, please make sure you set this correctly https://github.com/OpenDriveLab/TCP/blob/9ec4db0f0424801cdd607f1de930290830c5e88e/leaderboard/scripts/data_collection.sh#L2.

Mr-ChenSH, 2024-08-11:
Traceback (most recent call last):
  File "/home/cybersec/TCP-main/leaderboard/leaderboard/leaderboard_evaluator.py", line 34, in <module>
    from leaderboard.scenarios.scenario_manager import ScenarioManager
  File "/home/cybersec/TCP-main/leaderboard/leaderboard/scenarios/scenario_manager.py", line 25, in <module>
    from leaderboard.autoagents.agent_wrapper import AgentWrapper, AgentError
  File "/home/cybersec/TCP-main/leaderboard/leaderboard/autoagents/agent_wrapper.py", line 21, in <module>
    from leaderboard.autoagents.autonomous_agent import Track
  File "/home/cybersec/TCP-main/leaderboard/leaderboard/autoagents/autonomous_agent.py", line 17, in <module>
    from leaderboard.utils.route_manipulation import downsample_route
  File "/home/cybersec/TCP-main/leaderboard/leaderboard/utils/route_manipulation.py", line 22, in <module>
    from agents.navigation.global_route_planner import GlobalRoutePlanner
ModuleNotFoundError: No module named 'agents.navigation'

Mr-ChenSH, 2024-08-11:
export CARLA_ROOT=/home/cybersec/Work/Docker/carla/share/PythonAPI/carla
Even after adding it, there is still an error message

penghao-wu, 2024-08-11:
The structure of your carla directory looks a bit weird. Why is there a `share` directory? I think in your case, you need to set `CARLA_ROOT` to /home/cybersec/Work/Docker/carla/share

Mr-ChenSH, 2024-08-11:
sorry, I have already resolved this error

Mr-ChenSH, 2024-08-11:
Traceback (most recent call last):
  File "/home/cybersec/TCP-main/leaderboard/leaderboard/leaderboard_evaluator.py", line 350, in _load_and_run_scenario
    self._load_and_wait_for_world(args, config.town, config.ego_vehicles)
  File "/home/cybersec/TCP-main/leaderboard/leaderboard/leaderboard_evaluator.py", line 232, in _load_and_wait_for_world
    self.traffic_manager.set_synchronous_mode(False)
AttributeError: 'LeaderboardEvaluator' object has no attribute 'traffic_manager'
> Registering the route statistics
But another new error has occurred

Mr-ChenSH, 2024-08-11:
But another new error has occurred

penghao-wu, 2024-08-11:
Please check whether `traffic_manager` is successfully set here https://github.com/OpenDriveLab/TCP/blob/9ec4db0f0424801cdd607f1de930290830c5e88e/leaderboard/leaderboard/leaderboard_evaluator.py#L111

Mr-ChenSH, 2024-08-11:
[image]

Mr-ChenSH, 2024-08-11:
It's here, I haven't made any changes

penghao-wu, 2024-08-11:
I mean you can debug a little bit here to see whether it is successfully setup here in your case or it goes to the `except` branch.

Mr-ChenSH, 2024-08-12:
Can I run eval directly after running the train

Mr-ChenSH, 2024-08-12:
I encountered this issue while running eval
[image]

penghao-wu, 2024-08-12:
Have you modified the code? The line 5 in `TCP/model.py` should be https://github.com/OpenDriveLab/TCP/blob/9ec4db0f0424801cdd607f1de930290830c5e88e/TCP/model.py#L5

Mr-ChenSH, 2024-08-12:
yes，I modified it

Mr-ChenSH, 2024-08-12:
The current mistake is as follows：
[image]

penghao-wu, 2024-08-12:
Please first try the original code without your modifications to see whether it works.

penghao-wu, 2024-08-13:
I think it is related to your network issue.

Mr-ChenSH, 2024-08-13:
[image]
I downloaded it online, but now I'm encountering this issue. Do I need to include the dataset in the project during verification

Mr-ChenSH, 2024-08-13:
There is no dataset in this project

penghao-wu, 2024-08-13:
You don't need dataset for evaluation. Make sure you strictly follow the readme for the installation of carla and this repo.

Mr-ChenSH, 2024-08-13:
I have completed it, thank you very much！

closed 2024-08-13 by penghao-wu