OpenDriveLab/TCP #69: Data collection purpose

dmariaa, 2024-08-11:
Hi!! First of all thanks for your great work and open sourcing it...

I am just starting with your model and wondering about the purpose of the data collection procedure you have documented in the readme. You have published the links to the dataset too, so it can be downloaded and used for training. Is the data collection procedure only needed to recreate this dataset?

King regards and again thanks for your work...

penghao-wu, 2024-08-11:
Hi, if you just want to reproduce our model, you can directly use the uploaded dataset. The data collection procedure is provided so that people can collect more data or collect different data with certain modifications according to their needs.

Mr-ChenSH, 2024-08-14:
How to change the perspective of the car during verification